AmedeeBulle/pyrak811 #4: Add error messages

jezmck, 2019-06-11:
So that e.g. the following can be more descriptive.

    Traceback (most recent call last):
      File "/home/pi/lora/lora_node.py", line 20, in <module>
        lora.send('Hellora world')
      File "/usr/local/lib/python3.5/dist-packages/rak811/rak811.py", line 482, in send
        data
      File "/usr/local/lib/python3.5/dist-packages/rak811/rak811.py", line 226, in _send_command
        raise Rak811ResponseError(response[len(RESPONSE_ERROR):])
    rak811.rak811.Rak811ResponseError: -5

`rak811.rak811.Rak811ResponseError: -5` equates to "CODE_NOT_JOIN", which, while a little unnatural, is more useful.

AmedeeBulle, 2019-06-12:
Thank you for the report.

Note that you already can have the error message if you trap the exception yourself -- e.g:
```
>>> from rak811 import Rak811ResponseError
>>> try:
...     raise Rak811ResponseError(-5)
... except Rak811ResponseError as e:
...     print("[Errno  {}] {}".format(e.errno, e.strerror))
... 
[Errno -5] Not joined
```

But I will enhance the default message in the backtrace to produce:
```
>>> from rak811 import Rak811ResponseError
>>> raise Rak811ResponseError(-5)  
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
rak811.rak811.Rak811ResponseError: [Errno -5] Not joined
```

jezmck, 2019-06-13:
Thanks. I'm new to Python and appreciate your advice.

closed 2019-06-13 by AmedeeBulle

AmedeeBulle, 2019-06-13:
0.6.5 has been released on PyPI, just update your library and you will get more descriptive error messages.

```
sudo pip3 install --upgrade rak811
```

commit 631b2c4b (2019-06-13): Improve error message in backtrace when exception is raised.
Fixes #4.