AmedeeBulle/pyrak811 #5: Send to specific channel

varna9000, 2019-07-04:
How do you send a payload on specific channel number?

ryanteck, 2019-07-04:
Hi @varna9000 ,

Do you mean a specific LoRa Frequency (Such as only on 868.1 Mhz) or a specific lora port? (Instead of the default 1)

varna9000, 2019-07-04:
Yes, I think it's the frequency I'm looking for.

ryanteck, 2019-07-04:
Hi @varna9000 , Currently it isn't possible in the library but we'll get it implemented shortly.

Can I check which frequency you need yours to broadcast on?

varna9000, 2019-07-04:
I'm sorry it seems I made a mistake. I'm on a standard frequency :) Great library, thanks for your efforts.

AmedeeBulle, 2021-04-08:
As this issue is inactive for quite some time now, I am closing it.
If you feel it deserve additional attention, feel free to reopen it.

closed 2021-04-08 by AmedeeBulle