AmedeeBulle/pyrak811 #6: LoRaWAN adr implementation

demetz, 2019-07-13:
Hi,

I am confused with regards to the LoRaWAN adr implementation. My sender sits right next to the gateway, hence I would expect a spreading-factor of 7.

I am working on the commandline. After doing a `rak811 hard-reset`, `rak811 set-config [...]` and `rak811 join-otaa` I am able to send messages using `rak811 send`.

Both `JoinRequest` from the device as well as the `JoinAccept` from the gateway are with SF 7. Messages thereafter sent out by the device are sent out with SF 12.

`rak811 get-config adr` returns `on`, `rak811 get-config dr` returns `0`. `rak811 dr` also returns `0`.

Even after sending multiple messages, the spreading-factor remains unchanged at 12.

Is there anything else I need to do?  Is the `ADRACKReq` feature implemented?

AmedeeBulle, 2019-07-13:
There are a couple of things to look at...

Do not put your node close to the gateway, it has adverse effects on signal quality!

Also it takes quite some time before and ADR packet is sent to the node... Sending a downlink or a confirmed uplink can pass an ADR packet soon.
There are some edge cases as well with ADR and SF12...

ryanteck, 2019-07-13:
I think as @Amedee said most of those points are right. It can sometimes I
think take at least 30 packets for it to decide to reduce down.

AmedeeBulle, 2021-04-08:
As this issue is inactive for quite some time now, I am closing it.
If you feel it deserve additional attention, feel free to reopen it.

closed 2021-04-08 by AmedeeBulle

amedee, 2021-05-26:
I'm sorry to comment on an already closed issue, but why was I tagged? I am not involved with this project (as far as I know).

AmedeeBulle, 2021-05-27:
@amedee simply because somebody tagged you instead of tagging me ;)

Mvg