AmedeeBulle/pyrak811 #7: Add GPIO Commands

ryanteck, 2020-01-24:
We've had a request to add in the commands for the GPIO Pins. As I've been working on these for the micro:bit recently I'll work on this after TTN Conf

ryanteck, 2020-01-24:
Functions are:
Write to GPIO Pin
Read from GPIO Pin (Digital)
Read from GPIO Pin (Analogue)

panban-ux, 2020-12-03:
I was thinking can we trigger raspberry pi 4 GPIO's , lets say 5&6 to force a shutdown in this pHat under a downlink command? 
I'm using old firmware by the way

ryanteck, 2020-12-03:
@panban-ux do you mean use the GPIOs on the HAT to then trigger a shutdown of the Raspberry Pi?

I don't think that would be possible.

panban-ux, 2020-12-03:
Yes that's what i mean, using a specific downlink to the pHat.  Arent GPIO's used as inputs/outputs. I mean a way to trigger a high or low?

panban-ux, 2021-09-01:
Regarding my last post, Does a down link from TTN serves any purpose for this particular HAT? 
Lets say trigger a script on the Rpi
Im using the V2 firmware.

AmedeeBulle, 2021-09-01:
The GPIOs are of little use, the only thing you can do is set or query the state of the pin from the RPi.

They basically add a few GPIO pins to the Pi, without the flexibility of the native GPIO ports (you can't trigger events on state change, you have to poll).

I have implemented the GPIO interface for the V3 firmware if you are desperate to use these pins, but as I said usability is questionable...

AmedeeBulle, 2022-02-27:
Closing this issue, GPIO is supported with the V3 firmware for the RAK811 module; won't backport to V2.

closed 2022-02-27 by AmedeeBulle