AmedeeBulle/pyrak811 #8: Support new RAK firmware commands

ryanteck, 2020-05-07:
The next batch of pHATs will be using the new RAK firmware that has a different command structure.

I'll be doing the work over the next few days to do this. The current plan is:

Keep current version as is and branched off.

New version will be say 1.0.0 and both will be downloadable from pypi.

Tutorials then say to either download the version if the user knows. Or run a curl command to run the automated installer which will then detect which version is on the board and install that.

We'll write guides on how to upgrade existing nodes.

ryanteck, 2020-05-07:
RAK Firmware we'll go for is 2020-03-09.

AmedeeBulle, 2021-04-09:
Duplicate of #15

closed 2021-04-09 by AmedeeBulle