AmedeeBulle/pyrak811 #12: Confirmed send not working in my case

freeedom67, 2020-09-25:
Hey and thank you for you great lib,

i am using the pi phat lora node hat with a raspberry pi 3 and an own private lora network. I can send messages to the lora network without any problems from the raspberry pi using a python script. -as long as I use the unconfirmed send.

But when I use the confirm flag, the library always ends with a 'Rak811 timeout exception' after a long time waiting for the ACK I guess.  Somtetimes I can see the msg received by the gateway, sometimes I cant.

Do you maybe have an idea what the problem may be, or do you have another suggestion how to check, if a uplink was received correctly by my gateway? I cant see any logs of the gateway unfortunatly, i can just use it.

Thank you and best regards

ryanteck, 2020-09-25:
Hi @freeedom67 ,

From what I remember the exception is that it's not received a confirmation from the gateway, this could be while the gateway might pick up the packet the arrangement isn't strong enough signal for the gateway to transmit back and the node to pick it up.

Have you tried at a different data rate at all / Increasing the Spreading Factor?

freeedom67, 2020-09-25:
Hi @ryanteck , thak your for the fast answer!

Yes, i tried i with the lowest DR 0 , and  I can also receive the messages on the gateway. The point is, i think the whole downlink mechanism doesnt seem to work. I cant receive acks or downlinks on the lora node phat and I have no idea why. Could it be some configurations / channel settings which  I am missing? As i mentioned I dont use TTN.

Best Regards!

ryanteck, 2020-09-25:
Hi @freeedom67 ,

Sorry I missed that you're using a private network, it makes it slightly harder to debug as we don't really provide support if not using TTN.

Do you know what frequency it is trying to downlink on? And which region you're using to match up?

AmedeeBulle, 2020-09-25:
If you join with _OTAA_, your network should provide the right channel configuration.
However, if you use _ABP_ you might need to set RX2 to the correct channel and frequency depending on your region and network configuration...

freeedom67, 2020-09-25:
I will ask the network administrator and leave a comment as soon as i have an answer! Thank you four your help.

AmedeeBulle, 2020-09-25:
FYI, you can query your current RX2:

```shell
$ rak811 -v get-config rx2
0,869525000
```

In the above example downlink is expected on 869.525, datarate 0 (SF12 BW125 in the EU Frequency plan)

Similarly you can configure it to e.g. DR 3 (SF9 BW125 for EU):

```shell
$ rak811 -v set-config rx2=3,869525000
LoRaWan parameters set
$ rak811 -v get-config rx2
3,869525000
```

or in python:
```python
lora.set_config(rx2='3,869525000')
print(lora.get_config('rx2'))
```

freeedom67, 2020-10-13:
TY for your advice @AmedeeBulle ! Unfortunately I haven't been able to resolve the issue till now. The Lora network administrator told me, that the configuration of the private Lora network matches the EU868-870 standard. So the Lora node pHAT defaults configuration (0,869525000) should work, I guess.  

The strange thing is, that the Rak811 library always stuck when I : 

a) Try to join via OTAA
b) When I try to send a confirmed ABP Message

=> When I send unconfirmed via ABP everything works as expected. 

Do you maybe have another idea what the reason for this may be? Or where I could look to find a solution?

Another thing to the stuck of the Rak811 Library: Is it normal that it stuck when it doesn't receive an ACK (or at least thinks that it doesn't)? How do you usually handle retries then. What's the normal mechanism, if an ACK is lost. This should happen frequently when sending confirmed.

ryanteck, 2020-10-27:
Hi @freeedom67 ,

I've just been working on another one. 

This issue would be caused by the node either not receiving or the gateway not sending on the correct frequency the packet back.

Is the private network admin able to provide any logs of when the OTAA connection is attempted? (On TTN this is usually shown in 4 steps and shows the request to join, the join data and then the packets however I understand you're using a private network but hopefully those logs might show similar).

freeedom67, 2020-11-04:
hi @ryanteck ,

I am happy to tell that we found a solution, thank you for the advice. The problem was located in our private network server and is now fixed! 

Just another question about the confirmed send:

From time to time the rak811 library throws an error during send: 
**[Errno 6] rx2 timeout**

But the confirmation seems to work correctly. Did you ever experience this error msg?

ryanteck, 2020-11-04:
Hi @freeedom67 ,

It could be that the signal strength of the gateway back to the node isn't quite strong enough. I would investigate that and see if that potentially resolves it.

AmedeeBulle, 2021-04-08:
As this issue is inactive for quite some time now, I am closing it.
If you feel it deserve additional attention, feel free to reopen it.

closed 2021-04-08 by AmedeeBulle