AmedeeBulle/pyrak811 #15: Issues running RAK commands

willcurtis, 2021-03-02:
Hi Guys,

I have run in to an issue with the initial setup, it seems when running rak811 XXXX commands Im getting errors, for example, 

pi@raspberrypi:~ $ rak811 get-config dev_eui
RAK811 response error : 2: Unknown error

or 

pi@raspberrypi:~ $ rak811 reset module
Traceback (most recent call last):
  File "/usr/local/bin/rak811", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/rak811/cli.py", line 174, in reset
    lora.reset(Reset.Module)
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811.py", line 273, in reset
    self._send_command('reset={0}'.format(mode))
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811.py", line 233, in _send_command
    raise Rak811ResponseError(response[len(RESPONSE_ERROR):])
rak811.rak811.Rak811ResponseError: [Errno : 1] Unknown error

I have run through the basic setup found here - https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/

What am I doing wrong?

Im using RAK version 0.7.3

pi@raspberrypi:~ $ rak811 --version
rak811, version 0.7.3

The firmware version of the board is V3.0.0.14.H

Rgds, Will.

AmedeeBulle, 2021-04-08:
The (v3-firmware)[https://github.com/AmedeeBulle/pyrak811/tree/v3-firmware] development branch has LoRaWan support for the latest pHAT batch.

Support for LoRA P2P will be available soon.

AmedeeBulle, 2021-04-09:
[0.8.0.dev2](https://pypi.org/project/rak811/0.8.0.dev2/) version has now LoRa P2P support.

AmedeeBulle, 2021-04-10:
[0.8.0rc1](https://pypi.org/project/rak811/0.8.0rc1/) adds UART support.

Unless issues are found, it will be merged in the main branch soon.

Neogrey, 2021-04-27:
I unfortunately can' t get my lora phat to work, I followed the pi-supply guide and also get the error message:

```
pi@raspberrypi:~ $ rak811 get-config dev_eui
RAK811 response error : 2: Unknown error
```

I have seen this bug report and have updated rak811 library but still get the error message.

```
pi@raspberrypi:~ $ rak811 version
 V3.0.0.14.H

pi@raspberrypi:~ $ rak811 --version
rak811, version 0.8.0rc1
```

Shouldn't the problem be fixed with the new version? or am I doing something wrong?

AmedeeBulle, 2021-04-27:
When you have a V3 module, you need to use the `rak811v3` command.

See the [README](https://github.com/AmedeeBulle/pyrak811/tree/v3-firmware#readme) file from the `v3-firmware` branch for the the commands to use and pointers to examples.

The difference between V2 and V3 firmware are significant, we couldn't keep the same high level commands, so we have a new command line and a new API for V3.

The README should get you started.

(I'll merge the change to the main branch to make things clearer)

Neogrey, 2021-04-27:
> When you have a V3 module, you need to use the `rak811v3` command.
>
> See the [README](https://github.com/AmedeeBulle/pyrak811/tree/v3-firmware#readme) file from the `v3-firmware` branch for the the [...]

That was a quick response, using rak811v3 fixed the problem. Thanks a lot!

AmedeeBulle, 2021-04-27:
[rak811 0.8.0](https://pypi.org/project/rak811/0.8.0/) has been released, code merged to the main branch.

There is not code difference since 0.8.0rc1.

Closing this issue, for any bug in this new release, please file a new issue.

closed 2021-04-27 by AmedeeBulle