AmedeeBulle/pyrak811 #16: Error using rak811 commands

jakobfriis, 2021-03-04:
My new board does not work with more than version. 

Im using a fresh install on a raspberry pi 3, and im using this guide: https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/

After setting the uart, and installing rak811 using pip3, i can get the version from the hat:

pi@raspberrypi:~ $ sudo rak811 version
 V3.0.0.14.H

But getting any other parameter is not working

pi@raspberrypi:~ $ sudo rak811 hard-reset
pi@raspberrypi:~ $ sudo rak811 get-config ch_list
RAK811 response error : 2: Unknown error
pi@raspberrypi:~ $ sudo rak811 get-config dev_eui
RAK811 response error : 2: Unknown error
pi@raspberrypi:~ $ sudo rak811 get-config dev_addr
RAK811 response error : 2: Unknown error

Trying the sample fails (i know the x's must be replaced, but the initialization fails)

#!/usr/bin/env python3
from rak811 import Mode, Rak811

lora = Rak811()
lora.hard_reset()
lora.mode = Mode.LoRaWan
lora.band = 'EU868'
lora.set_config(dev_eui='xxxxxxxxxxxxxxxx',
app_eui='xxxxxxxxxxxxxxxx',
app_key='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx')
lora.join_otaa()
lora.dr = 5
lora.send('Hello world')
lora.close()

result:
pi@raspberrypi:~ $ python3 test.py
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    lora.mode = Mode.LoRaWan
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811.py", line 289, in mode
    self._send_command('mode={0}'.format(value))
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811.py", line 233, in _send_command
    raise Rak811ResponseError(response[len(RESPONSE_ERROR):])
rak811.rak811.Rak811ResponseError: [Errno : 1] Unknown error

jakobfriis, 2021-03-04:
pi@raspberrypi:~ $ pip3 list | grep rak811
rak811            0.7.3

hohenstaufen, 2021-03-24:
Is there any working solution to this problem?

AmedeeBulle, 2021-03-24:
I am working on an updated version.

AmedeeBulle, 2021-04-08:
The (v3-firmware)[https://github.com/AmedeeBulle/pyrak811/tree/v3-firmware] development branch has LoRaWan support for the latest batch.

Support for LoRA P2P will be available soon.

AmedeeBulle, 2021-04-09:
Closing as duplicate of #15

closed 2021-04-09 by AmedeeBulle