AmedeeBulle/pyrak811 #17: app_eui rejected if all zeros

mikethebee, 2021-04-07:
Apologies if this has been answered elsewhere, I did try a search.
TTN V3 stack suggests setting app_eui to all zeros for manual devices; I have older phat 2.0.3.0 modules that work fine with TTN V2 as there is a unique app_eui issued. rak811 join-otaa rejects an app_eui with all zero's as invalid. I have looked to see where this is checked, and it seems to be in the at+ code so may be outside the scope of this library. Has anyone else solved this other than using a fake app_eui. Thanks Mike

AmedeeBulle, 2021-04-07:
You are correct, there is no check in the library on the value of the `app_eui` parameter, it is rejected by the module.

Now I am a bit puzzled, how does the device connects with the app if you don't specify the `app_eui`?
(Apologies if this is a silly question, I am still using the V2 stack)

mikethebee, 2021-04-08:
Thanks for confirming that. I guess I need to look at the pisupply or RAK firmware as it the zero's are accepted ok on the RN2483.

To put it in as simpler way as I understand it, the V3 Stack uses a separate Join and App server instances. Depending on whether the client is >v1.1  or not it uses a (different) number of certs. (V1.1 is more secure but more restrictive) The cert keys and net:id determine the routing of packets rather than the `app_eui`. There are uses for the  `app_eui` in connection with non-TTN external join servers.

V3 also enables peering (roaming) that can expand the community network if private instances are allowed to forward packets to the community net in any particular area, as we plan to do with ours.

closed 2021-04-08 by mikethebee

AmedeeBulle, 2021-04-08:
FYI, I have a node here with the RAK811 V3 firmware and it does accept zero's:

```
$ rak811v3 -v set-config lora:app_eui:0000000000000000
Configuration done
$ rak811v3 get-config lora:status | grep -i 'appeui'
AppEui: 0000000000000000
```

There might be a possibility to upgrade the firmware of the RAK811 module.

jakobfriis, 2021-06-02:
Setting 0's is supported, but using them fails

root@raspberrypizero2:/home/pi# rak811v3 get-config lora:status
Work Mode: LoRaWAN
Region: EU868
MulticastEnable: false
DutycycleEnable: false
Send_repeat_cnt: 0
Join_mode: OTAA
DevEui: 60C5A8FFFE79858F
AppEui: 0000000000000000
AppKey: 468BEB780DED4881891187F1A8BBEF76
Class: A
Joined Network:false
IsConfirm: unconfirm
AdrEnable: true
EnableRepeaterSupport: false
RX2_CHANNEL_FREQUENCY: 869525000, RX2_CHANNEL_DR:0
RX_WINDOW_DURATION: 3000ms
RECEIVE_DELAY_1: 1000ms
RECEIVE_DELAY_2: 2000ms
JOIN_ACCEPT_DELAY_1: 5000ms
JOIN_ACCEPT_DELAY_2: 6000ms
Current Datarate: 5
Primeval Datarate: 5
ChannelsTxPower: 0
UpLinkCounter: 0
DownLinkCounter: 0
root@raspberrypizero2:/home/pi# rak811v3 -v join
**RAK811 response error 82: Invalid LoRa parameters**
root@raspberrypizero2:/home/pi#

AmedeeBulle, 2021-06-02:
Unfortunately there is nothing I can do in the scope of this repository as this comes directly from the RAK811 firmware.
You might want to check the [RAKwireless forum](https://forum.rakwireless.com/c/lpwan-modules-evb/rak811/47)...