AmedeeBulle/pyrak811 #19: Timeout Error on any rak811 commands

JoeKlimcak, 2021-05-19:
I recently bought the IoT LoRa Node pHAT, using with Raspberry Pi 3 1.2.

Whenever I try to run any rak811 commands, e.g. 
`rak811 -v reset lora`

I get:
```
    ...
    File "/usr/local/lib/python3.7/dist-packages/rak811/serial.py", line 156, in receive
        raise Rak811TimeoutError('Timeout while waiting for data')
rak811.serial.Rak811TimeoutError: Timeout while waiting for data
```

For reference, I did follow this [link](https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/), and set:
1. `sudo raspi-config`
2. select '3. Interface Options'
3. select 'P6 Serial Port'
4. 'Would you like a login shell to be accessible over serial?' -> No
5. 'Would you like the serial port hardware to be enabled?' -> Yes
6. `sudo nano /boot/config.txt`
7.  added `dtoverlay=pi3-miniuart-bt` to the end of the file
6. `sudo reboot`

Networking options were already configured before, i.e. it's connected to the internet.
But nothing seems to fix the above timeout error. Can you please tell me what I'm doing wrong?

AmedeeBulle, 2021-05-19:
If it is a new device it has most probably the RAK V3 firmware, which would explain the error you get.

PiSupply has not updated the instructions yet, but the [README](https://github.com/AmedeeBulle/pyrak811#readme) page of this project explains how to identify which firmware is installed on your pHAT and has examples for both versions.

JoeKlimcak, 2021-05-19:
Thanks for the prompt response.

Seems V3 has the same issue, ran:
1. `rak811v3 hard-reset`
2. `rak811v3 -v set-config lora:work_mode:0`
```
RAK811 timeout: Timeout while waiting for data
```

JoeKlimcak, 2021-05-19:
Thinking that I've only registered the device on TTN V2, so I'll probably need to register it on TTN V3 first? Will try that first.

AmedeeBulle, 2021-05-19:
I am still using TTN V2 as well, it should not matter.

Can you post the output of
```shell
ls -l /dev/serial*
ls -l /dev/ttyAMA*
rak811v3 -v -d version
 ```

In case the above succeeds, the output of
```shell
rak811v3 -v -d set-config lora:work_mode:0
```

I suspect something might be wrong with the serial port configuration. I don't have a Pi3 available to check (my test units are a Pi2 and a Pi4 ), but you can also test with `dtoverlay=disable-bt` instead of `dtoverlay=pi3-miniuart-bt` (reboot after the change).
Recent kernels don't need the `pi3-` prefix; see [UART Configuration](https://www.raspberrypi.org/documentation/configuration/uart.md).

JoeKlimcak, 2021-05-20:
Tried setting `dtoverlay=disable-bt`, outputs are below:

```
pi@raspberrypi:~ $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 7 May 19 14:17 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 May 19 14:17 /dev/serial1 -> ttyS0

pi@raspberrypi:~ $ ls -l /dev/ttyAMA*
crw-rw---- 1 root dialout 204, 64 May 19 14:17 /dev/ttyAMA0

pi@raspberrypi:~ $ ls -l /dev/ttyS*
crw-rw---- 1 root dialout 4, 64 May 19 14:17 /dev/ttyS0

pi@raspberrypi:~ $ rak811v3 -v -d version
DEBUG:rak811.serial:Serial initialized
DEBUG:rak811.serial:Sending: >at+version\r\n<
DEBUG:rak811.serial:Received: >OK2.0.3.0<
Traceback (most recent call last):
  File "/usr/local/bin/rak811v3", line 10, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1134, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1059, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1665, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1401, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 767, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/rak811/cli_v3.py", line 146, in version
    click.echo(lora.version)
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811_v3.py", line 435, in version
    return(self._send_command('version'))
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811_v3.py", line 251, in _send_command
    response = self._serial.receive()
  File "/usr/local/lib/python3.7/dist-packages/rak811/serial.py", line 156, in receive
    raise Rak811TimeoutError('Timeout while waiting for data')
rak811.serial.Rak811TimeoutError: Timeout while waiting for data

pi@raspberrypi:~ $ rak811v3 -v -d set-config lora:work_mode:0
DEBUG:rak811.serial:Serial initialized
DEBUG:rak811.serial:Sending: >at+set_config=lora:work_mode:0\r\n<
DEBUG:rak811.serial:Received: >ERROR-1<
RAK811 timeout: Timeout while waiting for data
```

AmedeeBulle, 2021-05-20:
> DEBUG:rak811.serial:Sending: >at+version\r\n<
> DEBUG:rak811.serial:Received: >OK2.0.3.0<

It is a V2 module, not a V3, so the `rak811` command should work.

The good thing is that your serial interface is properly configured.

Can you run the following:

```
rak811 hard-reset
rak811 -v -d version
rak811 -v -d set-config lora:work_mode:0
```

I'll try to re-install old modules on my test node today to verify we don't have a regression.
Apologies for the inconvenience.

JoeKlimcak, 2021-05-21:
Thanks Philippe, that seems to have worked. `rak811 -v join-otaa` times out but judging from #20, it means I'm not close enough to other gateways (albeit TTN coverage map suggests otherwise). I'll go test this out further in the field - thanks for the help.

closed 2021-05-21 by JoeKlimcak

AmedeeBulle, 2021-05-21:
Note that the internal antenna of the RAK811 module won't do miracles. It works for me as I have gateways in the house, but the range is not impressive.

It is better with the provided antenna, but you have to move the bridge from `INT` to `EXT` on the board, otherwise it won't be used!

JoeKlimcak, 2021-05-21:
Super helpful, Philippe, thank you. 

Can I double check with you, it seems both INT and EXT are connected right now as per pic below.

According to the [README](https://github.com/PiSupply/IoTLoRaRange/blob/master/IoT%20LoRa%20Raspberry%20Pi%20Node%20pHAT/README.md), it seems I have to unsolder INT to increase my EXT range?

[image]

AmedeeBulle, 2021-05-22:
Oh! I wasn't aware that boards were shipped with 2 inductors...

This is actually a worst case scenario in term of range. The good thing is that you can easily de-solder one without being worried of breaking it as it is not always easy to take these little things out ;)

So you definitely want to de-solder the `INT` inductor when you use the external antenna.

han2125, 2021-11-19:
@AmedeeBulle  hallo,I want to ask you ,how does rak868 connect  with ttn?