AmedeeBulle/pyrak811 #20: [Errno 99] LoRa join failed

mxzahedlim, 2021-05-19:
My project used RPi 3B+ pair with LoRa Node pHAT RAK811-H

i follow all instruction at README.md but i always get error
Traceback (most recent call last):
  File "/home/pi/Desktop/lora.py", line 11, in <module>
    lora.join()
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811_v3.py", line 501, in join
    self._send_command('join', timeout=self._event_timeout)
  File "/usr/local/lib/python3.7/dist-packages/rak811/rak811_v3.py", line 254, in _send_command
    raise Rak811ResponseError(response[len(RESPONSE_ERROR):])
rak811.rak811_v3.Rak811ResponseError: [Errno 99] LoRa join failed
>>> 
how to fix it , i dont know how to do.
Thanks you

AmedeeBulle, 2021-05-19:
The Error 99 -- `LoRa join failed` -- is returned by the RAK811 module when it can't connect to the LoRaWan network.

This typically happens when:
- `app_eui` and/or `app_key` are incorrect
- there is no LoRaWan gateway in your proximity

AmedeeBulle, 2021-09-22:
Closing sue as no feedback from OP

closed 2021-09-22 by AmedeeBulle