AmedeeBulle/pyrak811 #21: otaa_v3.py failing

BadenochBroadband, 2021-09-20:
The example file otaa_v3.py is failing.

On running send loop crashes without explanation. When the try block is removed the script crashes and shows the error 'rak811.serial.Rak811TimeoutError: Timeout while waiting for data'.

The command line utility rak811v3 errors with 'RAK811 timeout: Timeout while waiting for data'

Using the serial port directly uplink and downlinks work as expected. I am using a wrapper script to send commands easily - for reference it is attached as rakconnect.sh.txt.

I understand that v3 firmware is new and much different from v2. We would prefer to keep using v2 but it is impossible unless we find a way to downgrade firmware and / or choose which firmware we receive on ordering. I don't think that's possible to do. We are using IoT LoRa Node pHAT for Raspberry Pi (Multi Frequency).

I noticed there are some timeout variables set in the scripts but without understanding too much how they work I have had no luck in modifying them.

Attached are output from:

Running otaa_v3.py with no modifications
Running otaa_v3.py with the try block removed
Running the command line interface
Running using the serial port directly
 
[command-line-output.txt](https://github.com/AmedeeBulle/pyrak811/files/7196195/command-line-output.txt)
[crash-without-try-except-block.txt](https://github.com/AmedeeBulle/pyrak811/files/7196196/crash-without-try-except-block.txt)
[otaa_v3-output.txt](https://github.com/AmedeeBulle/pyrak811/files/7196197/otaa_v3-output.txt)
[serial-output.txt](https://github.com/AmedeeBulle/pyrak811/files/7196198/serial-output.txt)
[rakconnect.sh.txt](https://github.com/AmedeeBulle/pyrak811/files/7196209/rakconnect.sh.txt)

AmedeeBulle, 2021-09-21:
Thank you for the detailed issue report.

From what I can see, you always send in "confirm mode" which takes longer than expected and the library doesn't wait long enough...

I'll make some additional tests and see how to fix this.

A couple of questions:
- When using the serial port, how long (approximately) does it take before  you get the confirmation?
- Can you confirm it works in "unconfirmed mode"?

BadenochBroadband, 2021-09-21:
Hi, "confirm mode" may be a red herring because I just enabled it by following the command line guide example exactly. There is a command in the example to set the confirm mode on.

I normally work with uncofirmed mode. I made some tests and the error / problem is the same in both modes.

Strangely - at least it seems counterintuitive - the response time using the serial port is 5s in confirmed mode and 7s in unconfirmed mode. These figures are reliable, though slow seeming. I have made a few measurements and they are the same each time. I guess that the timeout is shorter than both those values.

AmedeeBulle, 2021-09-22:
I think that the unconfirmed takes a bit longer, because the module has to wait to check if there is any downlink available; so the total time is (send time + wait for downlink).

In the case of a confirmed send, it doesn't need to wait additionally for a possible downlink as it should be part of the send confirmation.

I do have similar total time with the CLI, but I don't have timeouts:
```
$ time rak811v3 -d -v send 'Hello'
DEBUG:rak811.serial:Serial initialized
DEBUG:rak811.serial:Sending: >at+send=lora:1:48656c6c6f\r\n<
DEBUG:rak811.serial:Received: >OK <
DEBUG:rak811.rak811_v3:No downlink
Message sent.
No downlink available.

real    0m6.772s
user    0m1.439s
sys     0m0.129s
```

Now I did a bit of debugging / tracing and there are two issues:

1. when running in confirmed mode,  we always get a confirmation downlink which will be empty as there is no actual downlink.  `lora.get_downlink()['data'].hex()` will raise an exception because `lora.get_downlink()['data']` will be an empty `str`, not bytes. I need to fix this bug [here](https://github.com/AmedeeBulle/pyrak811/blob/main/rak811/rak811_v3.py#L332-L338). If you run in unconfirmed mode, or if you test if `lora.get_downlink()['data']` is not empty before converting with `hex()`,  the send loop will run perfectly...
... unless you hit the next issue...

2. The send time is very consistent in my tests:
Unconfirmed:
```
Sending packets every minute - Interrupt to cancel loop
You can send downlinks from the TTN console
Sending packet
INFO:__main__:Sent message in 3.27 secs
Sending packet
INFO:__main__:Sent message in 3.27 secs
Sending packet
INFO:__main__:Sent message in 3.27 secs
Sending packet
INFO:__main__:Sent message in 3.27 secs
...
```
Confirmed:
```
Sending packets every minute - Interrupt to cancel loop      
You can send downlinks from the TTN console
Sending packet        
INFO:__main__:Sent message in 1.38 secs
Received                  
Sending packet            
INFO:__main__:Sent message in 1.37 secs
Received           
Sending packet              
INFO:__main__:Sent message in 1.38 secs
```

However, it happened once to me that the module started to respond consistently slower after switching between OTAA and ABP...

```
Joining
Sending packets every minute - Interrupt to cancel loop
You can send downlinks from the TTN console
Sending packet
INFO:__main__:Sent message in 7.96 secs
Received 
Sending packet
INFO:__main__:Sent message in 8.15 secs
Received 
Sending packet
INFO:__main__:Sent message in 7.98 secs
```
After a couple of `rak811v3 -v set-config lora:join_mode:1`, response time came back to _normal_...

I will make some additional tests and push a fix to address both issues.

closed 2021-09-22 by AmedeeBulle

AmedeeBulle, 2021-09-22:
rak811 0.8.4 is available on [PyPi](https://pypi.org/project/rak811/).

Let me know if it solves your issue.

Thanks

BadenochBroadband, 2021-09-22:
I upgraded to 0.8.4 and I confirm it solves our issue, both with the otaa_v3 example script and using the command line utility.

Thanks for your work and quick responses.

PR #22 (2021-09-22): Address timeout on send issues
Fixes #21