AmedeeBulle/pyrak811 #26: Registering new v3 TTN end device

hardillb, 2021-11-29:
Hi,

I'm trying to move my devices over to TTN v3 and to register a new device it's asking for "LoRaWAN version" and "Regional Parameter version".

Looking at the [datasheet](https://docs.rakwireless.com/Product-Categories/WisDuo/RAK811-Module/Datasheet/#description) it looks to be version 1.0.2 but I can't find any reference to REV A or REV B for the Regional Parameter version

A naive assumption would be REV A if a revision is not mentioned...

Any pointers?

AmedeeBulle, 2021-11-29:
I have migrated mines with `MAC V1.0.2` / `PHY V1.0.2 REV A` and they are all working...

... not sure though if it is the right setting because I just checked in the _export file_ I generated with `ttn-lw-migrate` as backup when I migrated and all devices there have `PHY V1.0.2 REV B` -- but that might be  a stub... 🤷

AmedeeBulle, 2021-11-29:
Actually, the screenshots from [Connecting to The Things Stack (TTN V3)](https://docs.rakwireless.com/Product-Categories/WisDuo/RAK811-Module/Quickstart/#the-things-stack-otaa-device-registration) are showing `PHY V1.0.2 REV B`; I probably should amend my config 😊

hardillb, 2021-11-30:
Thanks, the pointer to ttn-lw-migrate helped.

Only thing left is to work out how to get my Adafruit Feather with TinyLoRa to work with v3 (not looking hopeful)

closed 2021-11-30 by hardillb

AmedeeBulle, 2021-11-30:
> Only thing left is to work out how to get my Adafruit Feather with TinyLoRa to work with v3

I know nothing about TinyLoRa, but my Adafruit Feathers are working fine with LMIC...