AmedeeBulle/pyrak811 #31: Issues running RAK commands

AchrafAcheche, 2023-01-25:
Hi Guys,
I recently bought the IoT LoRa Node pHAT, using with Raspberry Pi 4 Model B 8GB RAM.
My OS Version : Raspbian GNU/Linux 10 (buster)
For reference, I did follow this [link](https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/),
I have run in to an issue with the initial setup, it seems when running rak811 commands I'm getting errors, for example,
[image]

[image]

Maybe my Phat and Paspberry not connected correctly  
[image]
[image]

AmedeeBulle, 2023-01-25:
If it is a new module, it has most probably a newer firmware (v3), and the instructions you follow are a bit outdated...

You can get your module firmware version with `rak811 version`.

If it is a V3, the README at the top of this repo should help you to proceed.

AchrafAcheche, 2023-01-25:
I tried ,but it's the same problem
[image]
[image]

AchrafAcheche, 2023-01-25:
rak811 version dosen't work
[image]

AmedeeBulle, 2023-01-27:
The script can open the device but doesn't get any response.

I don't have an RPi4 available for testing right now, but it looks like you have have Bluetooth assigned to `/dev/serial0` instead of the RAK811 module.

Can you completely disable Bluetooth and try again? 
( `dtoverlay=disable-bt` in `/boot/config.txt` should do the trick)

You should then have a single `/dev/serial*` device

AchrafAcheche, 2023-01-27:
Same problem Thank you anyway Philippe

AchrafAcheche, 2023-01-27:
One other question , does the phat have a led is on when it is powered ?

AmedeeBulle, 2023-01-29:
No, there is no led on the module...

I tested again on an RPi4, and this is what I have:

```
$ ls -l /dev/serial*
lrwxrwxrwx 1 root root 5 Nov  4  2021 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 Nov  4  2021 /dev/serial1 -> ttyAMA0
```

The  `rak811` library uses `serial0` (primary UART), as it is normally the one assigned to the GPIO header used by the module.
In my configuration, the primary UART  (`serial0`) is the _Mini UART_  (UART1 aka `ttyS0`).
This should be the default according to the [documentation](https://www.raspberrypi.com/documentation/computers/configuration.html#primary-and-secondary-uart).
In your first screenshot you have it as secondary...
(More reading about this: [Configuring UARTs](https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts))

I have attached my [config.txt](https://github.com/AmedeeBulle/pyrak811/files/10529468/config.txt), but the only relevant line is:
```
enable_uart=1
```

With all that, I have a successful connection to the module:
```
 $ rak811v3 -d version
DEBUG:rak811.serial:Serial initialized
DEBUG:rak811.serial:Sending: >at+version\r\n<
DEBUG:rak811.serial:Received: >OK V3.0.0.14.H<
V3.0.0.14.H
```

Hope this helps to solve your issue.

Limocello, 2024-03-26:
Hello There, 
I seem to face the same Issue with a Compute Module 4 and the Raspberry Pi zero LoRa HAT:
https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/
https://www.waveshare.com/wiki/CM4-IO-BASE-B#USB2.0
If I check for the Version (or do any other command)
[image]
The same Issue happens regardless of wether I use rak811 or rak811v3

I assume the issue comes from the fact that I'm using a HAT designed for a Raspberry Pi zero on a Raspberry Pi Compute Module 4, which from my knowledge has more UARTs and a different configuration (see attachement).
I have the same UART assignement:
[image]

I've disabled the serial login shell, tried switching the UART interfaces, adding the `dtoverlay=disable-bt`, reimaging the Raspberry Pi without any signs of success. I'm at my wits end either there's something grave I'm overseeing, my module is broken or there's an issue with configuring raspberry Pi 4's. 

One question remains, that could give me a lead: Basically I've also tried running `rak811v3 version` without the LoRa HAT attached and got the same timeout error. Whilst it completely makes sense, that this causes an error should it be the same error (rak811.serial.Rak811TimeoutError) or should there be a different error? Depending on this I might be able to conclude whether the Error lies within the software or if its actually my HAT that's not working.

I'd really appreciate any help even if its not with a compute module but a Raspberry Pi 4 as I've spent hours trying to fix this.
Thank you very much in advance
[config.txt](https://github.com/AmedeeBulle/pyrak811/files/14755839/config.txt)

billz, 2024-06-02:
I also encountered this but found a solution... sharing it here in the event it helps someone else. 

After disabling the serial console (or so I'd thought) using raspi-config, I ran into the same timeouts with `ttyAMA0`. I installed `lsof` to inspect the device file and found that `agetty` was using it:

```
$ ls -la /dev/serial0
lrwxrwxrwx 1 root root 7 Jun  1 06:27 /dev/serial0 -> ttyAMA0

$ sudo apt install lsof -y
$ sudo lsof | grep /dev/ttyAMA0
agetty     855            root    0u      CHR          204,64      0t0        132 /dev/ttyAMA0
```

Sure enough, the serial-getty@ttyAMA0 service was active:

```
$ sudo systemctl status serial-getty@ttyAMA0.service
● serial-getty@ttyAMA0.service - Serial Getty on ttyAMA0
     Loaded: loaded (/lib/systemd/system/serial-getty@.service; enabled-runtime; preset: enabled)
     Active: active (running) since Sun 2024-06-02 07:46:15 CEST; 6min ago
       Docs: man:agetty(8)
             man:systemd-getty-generator(8)
             https://0pointer.de/blog/projects/serial-console.html
   Main PID: 855 (agetty)
      Tasks: 1 (limit: 174)
        CPU: 8ms
     CGroup: /system.slice/system-serial\x2dgetty.slice/serial-getty@ttyAMA0.service
             └─855 /sbin/agetty -o "-p -- \\u" --keep-baud 115200,57600,38400,9600 - vt220

Jun 02 07:46:15 loranode2 systemd[1]: Started serial-getty@ttyAMA0.service - Serial Getty on ttyAMA0.
```
Stopping the service permits use of the serial port on ttyAMA0. Disabling it does _not_ prevent it from restarting on boot, though. This is due to systemd's use of [generators](http://www.freedesktop.org/wiki/Software/systemd/Generators/). The solution in this case is to mask the service to prevent it from ever being enabled:

```
$ sudo systemctl mask serial-getty@ttyAMA0.service
```

After doing this I had no issues using rak811 or rak811v3.

Environment:

Pi Zero 2 W with Debian 12/Raspberry Pi OS (64-bit) Lite bookworm.
```
$ uname -a
Linux loranode2 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
```

BryanNsoh, 2024-07-10:
You are much appreciated billz! I wrote a script anyone can run and it does everything recommended across all the issues and tells you what the most likely issue is. hope someone finds it helpful. also thanks AmedeeBulle for diligent replies!!

Here's the script that finally solved my problem. It performs all the checks and fixes recommended throughout this thread:

```bash
#!/bin/bash

# LoRa pHAT Fix Script
# This script performs checks and fixes for common LoRa pHAT issues

# Function to log messages
log_message() {
    echo "$1"
}

# Check and fix UART configuration
fix_uart_config() {
    if ! grep -q "enable_uart=1" /boot/config.txt; then
        log_message "Adding enable_uart=1 to /boot/config.txt"
        echo "enable_uart=1" | sudo tee -a /boot/config.txt
    fi

    if ! grep -q "dtoverlay=disable-bt" /boot/config.txt; then
        log_message "Adding dtoverlay=disable-bt to /boot/config.txt"
        echo "dtoverlay=disable-bt" | sudo tee -a /boot/config.txt
    fi
}

# Stop and disable serial-getty service
fix_serial_getty() {
    log_message "Stopping and disabling serial-getty@ttyAMA0.service"
    sudo systemctl stop serial-getty@ttyAMA0.service
    sudo systemctl disable serial-getty@ttyAMA0.service
    sudo systemctl mask serial-getty@ttyAMA0.service
}

# Main execution
log_message "Starting LoRa pHAT Fix Script"

fix_uart_config
fix_serial_getty

log_message "Fixes applied. Please reboot your Raspberry Pi."
log_message "After reboot, try running 'rak811 -v -d version' to check if the issue is resolved."
log_message "If problems persist, please check physical connections and antenna configuration."

# Prompt for reboot
read -p "Would you like to reboot now? (y/n) " -n 1 -r
echo
if [[ $REPLY =~ ^[Yy]$ ]]
then
    sudo reboot
fi

```

To use this script:

1. Save it to a file named `lora_phat_fix.sh`
2. Make it executable: `chmod +x lora_phat_fix.sh`
3. Run it with sudo: `sudo ./lora_phat_fix.sh`

This script will:
- Add necessary UART configurations to /boot/config.txt if they're missing
- Stop, disable, and mask the serial-getty service
- Prompt you to reboot, which is necessary for changes to take effect

After running this script and rebooting, most common LoRa pHAT issues should be resolved. If anyone is still having trouble, they should check their physical connections and antenna configuration.