AmedeeBulle/pyrak811 #32: RAK811 library crashes on Bookworm O/S (kernel v6.6.45)

timiman, 2024-08-15:
Trying to hard reset a pHAT RAK811 v3 LoRa Node with a RPi 5 running Bookworm O/S, gets me to the following error:

`
rak811v3 hard-reset
Traceback (most recent call last):
  File "/home/####/.local/bin/rak811v3", line 5, in <module>
    from rak811.cli_v3 import cli
  File "/home/####/.local/pipx/venvs/rak811/lib/python3.11/site-packages/rak811/cli_v3.py", line 26, in <module>
    from .rak811_v3 import Rak811
  File "/home/####/.local/pipx/venvs/rak811/lib/python3.11/site-packages/rak811/rak811_v3.py", line 26, in <module>
    from RPi import GPIO
ModuleNotFoundError: No module named 'RPi'
`

To be able to install the rak811 package in python 3.11, I've install pipx first to create a virtual-environment.
`
apt install pipx
pipx install rak811
` 

Should the error be by the obsolete way of GPIO access that has been changed in Bookworm O/S (latest kernels) or due to the virtual environment RPi package is not installed and thus not found when trying to import it ?

AmedeeBulle, 2024-08-19:
Do you have the `python3-rpi-lgpio` packages installed?

I haven't tested on recent kernels, but `rpi-lgpio` should offer compatibility for programs using the legacy `RPi.GPIO` interface...

timiman, 2024-08-19:
Thank you for the answer. I've tried just now but it seems that the major change has to do with the Raspberry Pi 5 itself. The GPIO is now placed in a separate chip, so libraries using GPIO have to be updated for it to work.

AmedeeBulle, 2024-08-20:
Right, so we have two changes to cope with...

Unfortunately, while I have all other RPi models, I don't own a version 5, it might take some time before I can debug this.

Having said that, the GPIO port is only used to perform a hard reset of the module, and I don't think it is critical for the V3 firmware.
If you have the serial port properly configured (that is: having the module bound to `/dev/serial0`), it should work without the use of the GPIO stuff.

timiman, 2024-08-21:
I've double-checked that the serial port is closed for login shell and enabled as hardware.
`rak811v3 get-config lora:status` and `rak811v3 -v -d version` stop due to timeout.
In /boot/firmware/config.txt I've added: 
`dtoverlay=disable-bt` and
`enable_uart=1`

I do have an antenna connected to the pHat (I have read that booting without an antenna connected possibly will kill the board, if true).

And last, `dmesg | grep tty` gives me:
`[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe  smsc95xx.macaddr=2C:CF:67:22:BA:41 vc_mem.mem_                                                   base=0x3fc00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=e00ad1d9-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymou                                                   th.ignore-serial-consoles cfg80211.ieee80211_regdom=GR
`
`[    0.000159] printk: console [tty1] enabled`
`[    0.010783] 107d001000.serial: ttyAMA10 at MMIO 0x107d001000 (irq = 15, base_baud = 0) is a PL011 rev2`
`[    0.359141] 107d50c000.serial: ttyS0 at MMIO 0x107d50c000 (irq = 33, base_baud = 6000000) is a Broadcom BCM7271 UART`
`[    0.538337] 1f00030000.serial: ttyAMA0 at MMIO 0x1f00030000 (irq = 125, base_baud = 0) is a PL011 AXI`

Maybe the pHat is not recognized ? Have I missed anything else ?

AmedeeBulle, 2024-08-30:
I have done a fresh install of the latest Raspberry Pi OS (Bookworm) on a RPi 4B and it all works as expected (using `rpi-lgpio`)

The hat is detected properly:
```
$ for i in /proc/device-tree/hat/*;do echo "$i: $(cat -v $i)"; done
/proc/device-tree/hat/name: hat
/proc/device-tree/hat/product: IoT LoRa Node pHAT
/proc/device-tree/hat/product_id: 0x0006
/proc/device-tree/hat/product_ver: 0x0101
/proc/device-tree/hat/uuid: <redacted>
/proc/device-tree/hat/vendor: Pi Supply
```

Serial is enabled / login on serial disabled
```
$ dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:AD:04:55 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=eb6044a2-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=BE
[    0.000328] printk: console [tty1] enabled
[    1.528481] fe201000.serial: ttyAMA1 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.528726] serial serial0: tty port ttyAMA1 registered
[    1.531624] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 37, base_baud = 62500000) is a 16550
[    3.965483] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
$ ls -l /dev/seria*
lrwxrwxrwx 1 root root 5 Aug 30 14:28 /dev/serial0 -> ttyS0
```

Lora module is responding:
```
$ rak811 version
 V3.0.0.14.H
$ rak811v3 get-config lora:status
Work Mode: LoRaWAN
Region: EU868
MulticastEnable: false
DutycycleEnable: false
Send_repeat_cnt: 0
Join_mode: OTAA
DevEui: <redacted>
AppEui: <redacted>
AppKey: <redacted>
Class: A
Joined Network:false
IsConfirm: unconfirm
AdrEnable: true
EnableRepeaterSupport: false
RX2_CHANNEL_FREQUENCY: 869525000, RX2_CHANNEL_DR:0
RX_WINDOW_DURATION: 3000ms
RECEIVE_DELAY_1: 1000ms
RECEIVE_DELAY_2: 2000ms
JOIN_ACCEPT_DELAY_1: 5000ms
JOIN_ACCEPT_DELAY_2: 6000ms
Current Datarate: 5
Primeval Datarate: 5
ChannelsTxPower: 0
UpLinkCounter: 0
DownLinkCounter: 0
```

One thing for sure, the OS version (bookworm) is not the culprit.
It is either something linked to the RPi 5 architecture -- but I can't test that -- or your module.

timiman, 2024-08-30:
OK. I also see the hat is detected in my RPi 5.

>  $ for i in /proc/device-tree/hat/*;do echo "$i: $(cat -v $i)"; done
/proc/device-tree/hat/name: hat
/proc/device-tree/hat/product: IoT LoRa Node pHAT
/proc/device-tree/hat/product_id: 0x0006
/proc/device-tree/hat/product_ver: 0x0101
/proc/device-tree/hat/uuid: <removed>
/proc/device-tree/hat/vendor: Pi Supply

But it seems that in my case it is using /serial1 and not /serial0.
Also the baud rate seems different, unless the ttyS0 is taken by another device (?)

>  dmesg | grep tty
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe  smsc95xx.macaddr=2C:CF:67:22:BA:41 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=e00ad1d9-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=GR
[    0.000158] printk: console [tty1] enabled
[    0.010684] 107d001000.serial: ttyAMA10 at MMIO 0x107d001000 (irq = 15, base_baud = 0) is a PL011 rev2
[    0.359276] 107d50c000.serial: ttyS0 at MMIO 0x107d50c000 (irq = 33, base_baud = 6000000) is a Broadcom BCM7271 UART
[    0.537779] 1f00030000.serial: ttyAMA0 at MMIO 0x1f00030000 (irq = 125, base_baud = 0) is a PL011 AXI
[    1.772163] systemd[1]: Created slice system-getty.slice - Slice /system/getty.

> $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 8 Aug 29 09:04 /dev/serial0 -> ttyAMA10
lrwxrwxrwx 1 root root 5 Aug 29 09:04 /dev/serial1 -> ttyS0

> It is either something linked to the RPi 5 architecture -- but I can't test that -- or your module.
It should be the GPIO access changes on RPi 5.

I hope you would step upon a RPi5 soon and give it a go with the LoRa pHat.

Also, does the link below give some info regarding the issue ?
[https://forums.raspberrypi.com/viewtopic.php?t=362821](url)

timiman, 2024-08-30:
Just a quick hard-rest I did try:

> $ rak811v3 hard-reset
Traceback (most recent call last):
  File "/home/jedipi/rak811_env/bin/rak811v3", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/rak811/cli_v3.py", line 72, in hard_reset
    lora.hard_reset()
  File "/home/jedipi/rak811_env/lib/python3.11/site-packages/rak811/rak811_v3.py", line 193, in hard_reset
    GPIO.setup(RESET_BCM_PORT, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address

And regarding the GPIO access on RPi 5, I've found this below mentioning that RPi and rpi-lgpio are not compatible:
[https://forums.raspberrypi.com/viewtopic.php?t=369089](url)

AmedeeBulle, 2024-08-30:
> But it seems that in my case it is using /serial1 and not /serial0. Also the baud rate seems different, unless the ttyS0 is taken by another device (?)

You need to ensure it isn't used as linux console and there is no login prompt on it (`agetty` running on that port)

> Also, does the link below give some info regarding the issue ? [https://forums.raspberrypi.com/viewtopic.php?t=362821](https://forums.raspberrypi.com/viewtopic.php?t=362821)

Yes, you definitely need serial on GPIO 14/15

Re. `lgpio` and the hard reset, it is a non issue for the LoRa modules running V3 firmware.
We need `lgpio` as it is a dependency; but it is only used for the hard reset, and the hard reset is only needed for the old V2 firmware.

timiman, 2024-08-30:
Ok with lgpio and hard-reset. It is not needed in v3.

So, following [https://github.com/AmedeeBulle/pyrak811/issues/31](url)
I see that services on ttyAMA0 and ttyAMA10 are dead and agetty is not running.

> sudo systemctl status serial-getty@ttyAMA10.service
○ serial-getty@ttyAMA10.service
     Loaded: masked (Reason: Unit serial-getty@ttyAMA10.service is masked.)
     Active: inactive (dead)
>  sudo systemctl status serial-getty@ttyAMA0.service
○ serial-getty@ttyAMA0.service
     Loaded: masked (Reason: Unit serial-getty@ttyAMA0.service is masked.)
     Active: inactive (dead)

I also have enable_uart=1 and dtoverlay=disable-bt on /boot/firmware/config.txt (edited manually as the script in link above uses before Bookworm filepath (/boot/config.txt).

How to tackle the timeout I get with `rak811v3 get-config lora:status` ?
Is there a method or command to determine if GPIO@14/15 are bind to serial or not ?
Can I provided you of any info from RPi 5, in case it can be helpful to solve the timeout I get ?

timiman, 2024-09-28:
Is there any progress regarding this issue ?
I'm still willing to help by providing info from my side, because sooner or later more users will face this issue with Raspberry Pi 5 and IoT LoRa Node pHAT.

vovabush, 2024-11-15:
Для того, чтобы IoT LoRa Node pHAT работала на Raspberry Pi 5 в /boot/firmware/config.txt в конце должно быть написано:
```
[all]
dtparam=uart0=on
```
Помимо этого нужно удалить старую библиотеку для GPIO 
```
sudo apt remove python3-rpi.gpio
```
и установить новую:
```
git clone https://github.com/waveform80/rpi-lgpio
cd rpi-lgpio
sudo python<python3> setup.py install
```
А в исходниках в serial.py нужно поменять ```PORT='/dev/serial0'``` на ```PORT='/dev/ttyAMA0'```.
После этого установить проект через `sudo python<python3> setup.py install` и всё будет ок:
[image]

timiman, 2024-11-15:
Thanks vovabush.
I'm now able to see and communicate with the LoRa node from TTN.

closed 2024-11-15 by timiman